ghi-electronics/codal-brainpad #40: Need to figure out which concrete impl to use for I2C

Arke64, 2018-05-04:
codal::i2c is just a base class.

pelikhan, 2018-05-08:
@VictorDuma @Arke64 could you look into creating a PR for codal-mbedos? What is the status here?

Arke64, 2018-05-08:
@pelikhan do you mean a PR to update lancaster-university/codal-mbedos I2C to match what was done in codal-mbed, as mentnioed in lancaster-university/codal-core#26?

pelikhan, 2018-05-08:
Yes similar to the PR that James mentionned.

Arke64, 2018-05-10:
We'll take a look and see what we can do

Arke64, 2018-05-10:
@pelikhan see lancaster-university/codal-mbedos#12

Arke64, 2018-05-10:
Done in above PR, will keep using `codal::_mbed::I2C`

closed 2018-05-10 by Arke64

pelikhan, 2018-05-10:
Do you have a new tag?

Arke64, 2018-05-10:
We didn't have to change anything in BrainPad, we were already using `codal::_mbed::I2C`. We'll get a new tag as soon as we finish the display format change.